bambulab/BambuStudio #5852: Bambu AMS Filaments can't be changed

UBaer21, 2025-01-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Printer

P1S AMS

### How to reproduce

If you calibrate Flow the new Flow factor is saved to a new filament,
But that Filament can't be set in AMS. As only the system filaments are set when RFID tag is read.

### Actual results

Sync AMS will revert to the default bambu filaments not the calibrated.
So having 16 Bambu filaments means you need to change them all the time, because if you change printer those filament list get messed up for no reason.

### Expected results

Calibration need to be stored in the system filament or at least ALL filaments should be able to be changed on AMS - so i can select the calibrated version.

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2025-02-05:
Could you please simply take some screenshots or record a video to briefly describe your problem?

UBaer21, 2025-02-05:
[image]

[image]

[image]

As you can see I can set my Filaments to be the calibrated profiles. But the RFID Bambu Filaments can't be changed - so i can not use the calibrated profile. When Sync the AMS filaments it will always use the system presets - so i have to manually switch all slots to the calibrated flow profiles. The only option at the moment is to open the Spool and remove the RFID tags. (or clip the AMS RFID sensor)

zhimin-zeng-bambulab, 2025-02-05:
For Bambu machines, the k value used is not the k value in the filament preset, but the k value set for each slot in the device page. Usually, you can manually set the k value for each slot (once set, this slot will always use this k value), or set the k value through automatic calibration.

UBaer21, 2025-02-05:
This is not about the k value but all other filament settings like cooling or scarf, temp and flow.

When i print with non bambu filament and bambu filament - they have different profiles because i can not change the bambu default profile. 

If i make a flow rate calibration (not dynamic k value), BS will safe a new calibrated filament profile - but this can't be set for the AMS so i have to select this manually every time.

zhimin-zeng-bambulab, 2025-02-05:
You can directly set each filament to your customized filament preset on the prepare page. The filament settings(like cooling or scarf, temp and flow) you mentioned only affect the slicing results. When sending to print, map it to the corresponding ams material.

UBaer21, 2025-02-05:
If you have multiple AMS - and you switch printer/nozzles, you have to sync the filaments from AMS every time again, because they get messed up. 
So Yes at the moment i have to manually select all Bambu filaments, while all non bambu just automatically select the correct profile from the AMS.

If i made a calibrated profile for my bambu filament i like to select this on the AMS so when i sync the AMS filaments it chooses the correct calibrated profile.  But it will only use the system preset which can't be changed.

zhimin-zeng-bambulab, 2025-02-05:
This is a good idea and we will consider optimizing it in the future. Thanks for this idea!