bambulab/BambuStudio #5853: Auto Refill mixes different colours

Goger-TKL, 2025-01-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: AMD 5800X3D
GPU: NVIDIA RTX 3080

### Printer

P1S

### How to reproduce

I have loaded 4 Bambu spools with different colours in the AMS. I make a print and a colour runs out. Since there is no matching colour I expect to ask for a refill but instead printig is continued with a different colour. This has not happened with older Bambu Studio Versions.

[image]

[image]

[image]

### Actual results

Printing continues with wrong colour.

### Expected results

Ask the user for a refill of the empty spool.

### Project file & Debug log uploads

[QTip Box+Lid V1.zip](https://github.com/user-attachments/files/18545349/QTip.Box%2BLid.V1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

jeremywillans, 2025-02-09:
I am also seeing this issue, and its even including an empty slot in the PLA Group.

[image]

Same BS version as OP, P1S with 2x AMS

XinZhangBBL, 2025-02-12:
Thank you for your feedback.
Also thanks @[jeremywillans](https://github.com/jeremywillans) for  your support
It has been identified as a firmware error. The issue will be fixed in the next firmware version

Goger-TKL, 2025-02-12:
Thank you for the Update. Looking forward to the new Firmware.

williams3dprints, 2025-02-16:
I just noticed the same problem with mine, is there an approximate timeframe to expect the release of the next firmware update?

gaolegao-lx, 2025-03-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

jibreil, 2025-04-30:
It looks like this may have been fixed in recent firmware updates.

Upgraded my P1S to 01.08.00.00 and AMS to 00.01.06.62 today and it seems to be resolved.

Unfortunately I don't have any open spools of the same colour at the moment to test properly, but previously I was getting 2 different colours in the `Auto-refill` dialogue, now it just shows none to auto-refill.

[image]

closed 2025-05-07 by BoBoYu-BamBu