bambulab/BambuStudio #5855: Stop automatically dropping imported objects to bed

ansonl, 2025-01-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Bambu Studio currently auto drops objects to the bed unless they are imported as a single nested object. 

I have multiple pieces in a file that need to be split and printed on different beds. Bambu Studio only lets them be imported as all separate objects or a single object. 

- Importing as separate objects drops them all to bed
- Splitting the single object in BS drops them to bed before I can Merge them into smaller collections.

In both cases the positioning of objects is lost.

Bambu Studio should add an option to disable automatically dropping objects to bed on import and splitting so that users can retain the object positioning.

### Printer

P1S

### How to reproduce

1. Import file with multiple objects A B C D that are not sitting on the bed
2. Try to regroup them into a grouping (A B) (C D)

### Actual results

The object positioning is lost before you can regroup them. 

### Expected results

Object positioning should be maintained when importing separately or after splitting.

### Project file & Debug log uploads

[CA-sqrt-dual-transparent-v2-hex200scale-test.zip](https://github.com/user-attachments/files/18548800/CA-sqrt-dual-transparent-v2-hex200scale-test.zip)

No log file needed. 

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-02-01:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
(https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

 
[image]
Can you provide the original STL or obj files？I can understand some of what you said, but I haven't fully understood it yet. Could you please provide a more detailed explanation? Thank you

ansonl, 2025-02-01:
@Haidiye00, I colored the separate objects in the attached sample file for the issue to make it more clear.

Screen recording showing the default imported grouping and drop to bed on split: [Animation.zip](https://github.com/user-attachments/files/18629781/Animation.zip)

In the model shown the green pieces are two separate objects that are positioned next to each other and each green object has a blue object that is positioned at a specific height in the 3MF. Making 4 objects total where 2 objects should be grouped together.

The first screenshot is what happens when I choose to not import as a single group with multiple parts. In the next screenshot I moved one of the green objects so you can see that there is a blue object hidden inside it that was dropped down to the build plate when imported (which is not what I want)

[image]

[image]

In the third screenshot, I import the same single 3MF file with 4 objects and select Yes to import as a grouped object. The blue objects are now kept at the correct height and not dropped to the bed. However I cannot separate and regroup the bottom green and blue objects without using the Split feature. Once the the single group is split I cannot regroup them at the correct position because the pieces are dropped to bed. Splitting leads to screenshot 1 instantly.

[image]

Ideally the final objected get grouped like the below so that I can print each group of green and blue objects separately and scale them, etc.

[image]

When the objects are not dropped to bed on split and I can regroup them correctly, I can scale and move each pair of objects separately to get a result like below:

[image]

Haidiye00, 2025-02-02:
Hello, thank you for your GIF. I have a general understanding of your needs and will evaluate and consider how to improve it. Thank you for your suggestion~

Haidiye00, 2025-04-29:
Hello, we have already resolved the issue internally and expect to bring it into the next version press conference.

hotellonely, 2025-06-03:
This is massive, I can't even imagine that this got solved on BS earlier on OS lmao