bambulab/BambuStudio #5856: Bambu Studio - Cannot read any dialog boxes, hints and instructions

slawomirrzeszow, 2025-01-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

When using Bambu Studio and trying to either slice the model or manually color the support

### Actual results

[image]
Cannot see text in dialog boxes. Text either dissapeared or the dialog box is almost completely black and it is hard to read anything

### Expected results

Normal readible dialog boxes

### Project file & Debug log uploads

[Halloween_Balance_Game.zip](https://github.com/user-attachments/files/18549727/Halloween_Balance_Game.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

lanewei120, 2025-02-01:
Hello, @slawomirrzeszow 

thanks for reporting this issue

is this issue 100% reproduced?
and is this issue found on previous versions?

guanyun-gudujian, 2025-02-01:
@slawomirrzeszow 
Currently we use compresed texture BC5 for font rendering, which may not supported on some devices. We will fix it in the near future.
Thanks for your report

Haidiye00, 2025-02-01:
[image]
Hello, if possible, could you please help take a screenshot of your graphics card model from the device manager? Thank you, it's convenient for us to reproduce this issue