bambulab/BambuStudio #5858: File Repair hangs and Bambu Studio eventually crashes with version 1.10.1

philchfree, 2025-01-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Import step file with non-manifold errors
Click Repair at the prompt

### Actual results

Program hangs and finallly crashes
All is fine with version 1.9.7.52 of Bambu Studio

### Expected results

File should repair

### Project file & Debug log uploads

x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-05:
Thank you for your feedback. Please provide the log and dmp files for us to analyze. Additionally, could you kindly provide the 3mf file related to the problem so that we can reproduce and locate the issue

philchfree, 2025-02-05:
Hi, It was just due to my STEP file having some embedded micro objects. It was a huge file and this was certainly due to remnants of extrusions while designing it. As soon as I deleted these, the file repaired as expected. Previous version of Studio might have been more compliant of these but as I know the issue now, I will check this when file does not repair properly in the future.

JohnSmithCYM, 2025-02-06:
Thanks for your feedback, and you still can offer us the step for further test, we just want to make sure the performance of STEP import same with the old version of Studio.

philchfree, 2025-02-06:
Hi, thank you for this. I uploaded a 3mf test file on which I experience the issue and the debug log (although you don't really need it, just try repairing with BS 1.9.7.52 and 1.10.1.50 and you will see the difference)

[Test File.zip](https://github.com/user-attachments/files/18688221/Test.File.zip)

I know now it's just because of the small object artifact that is present in the file but 1.9.7.52 did not seem to mind :).

Thanks and best regards

MackBambu, 2025-03-25:
@philchfree Enabling the "split compound and compsolid into multiple objects" option may filter out some incorrect small faces, but removing the erroneous faces directly is a better approach.