bambulab/BambuStudio #5860: Bambu Studio Disconnecting from Bambu P1S Printer

dhartun1, 2025-01-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	AMD Ryzen 9 5900X 12-Core Processor, 3701 Mhz, 12 Core(s), 24 Logical Processor(s)
OS Name	Microsoft Windows 11 Pro
Name	NVIDIA GeForce RTX 3090

### Printer

Bambu P1S

### How to reproduce

**Bug Report: Bambu Studio Disconnecting from Bambu P1S Printer
Summary:**
Bambu Studio on desktop frequently disconnects from the Bambu P1S printer after a recent software update. This issue did not occur previously and seems unrelated to any network or hardware changes.

**Steps to Reproduce:**

1. Launch Bambu Studio on a desktop connected via Ethernet.
2. Attempt to connect to the Bambu P1S printer.
3. Observe that the connection does not establish automatically or drops intermittently.

**Additional Information:**

- Printer and phone are connected to a 2.4GHz Wi-Fi network, while the desktop is connected via Ethernet.
- No hardware or network changes have been made aside from the software update.
- The issue only started after updating Bambu Studio.

**Environment Details:**

Printer Model: Bambu P1S
Desktop Connection: Ethernet
Phone Connection: Wi-Fi (2.4GHz)
Printer Connection: Wi-Fi (2.4GHz)
Bambu Studio Version: 1.10.1.50 (most recent version according to app at time of submitting)
Operating System: Windows 11

**Request:**
Please investigate the root cause of this connectivity issue introduced in the latest software update and provide a resolution or workaround.

### Actual results

**Observed Behavior:**

- The desktop app fails to reconnect to the printer automatically.
- Opening the Bambu Handy app on a phone (connected to a 2.4GHz Wi-Fi network) and selecting "Printer" under the "Devices" tab triggers the desktop app to reconnect to the printer.

### Expected results

**Expected Behavior:**

- Bambu Studio should maintain a stable connection with the printer without manual intervention via the phone app.

### Project file & Debug log uploads

[56、A1双双一体多色.zip](https://github.com/user-attachments/files/18550553/56.A1.zip)
[log.zip](https://github.com/user-attachments/files/18550554/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-02-06:
@dhartun1 Thanks for your feedback, and I have tested my P1S with V1.10.1.50 
- Over 40 miniutes it didn't occur your problem.
- Then I encountered your situation due to wifi is not steady.
So could offer a P1S log for us? You can get the log by https://wiki.bambulab.com/en/X1/troubleshooting/how-to-upload-log.

walterwongbbl, 2025-02-07:
@dhartun1 Can you use the same network, such as only using the computer's network?