bambulab/BambuStudio #5863: Slicer creates floating supports

jeffreylanters, 2025-01-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.1.1 (24B91)

### Additional system information

Apple M1 Pro

### Printer

Bambu Lab A1 Mini

### How to reproduce

It randomly happens when slicing models using the 0.2 nozzle with the default 0.10mm High Quality preset.

### Actual results

Supports start out of nowhere without any underlaying structure.

[image]

[image]

[image]

### Expected results

For supports to not start out of nowhere.

### Project file & Debug log uploads

Cannot share file due to purchased models.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NothFie, 2025-02-05:
@jeffreylanters Thanks for your report. Could you please take a screenshot of your parameter settings(or describe which parameters have been modified) and show it to me? It helps us find out the cause of this issue.

MartinDrost, 2025-02-07:
I'm encountering the same problem in my print (Bambu Studio v1.10.1.50). I was using overall custom settings but I have been able to reproduce it by also using the 0.10mm High Quality preset for the 0.2 nozzle and just entering my custom support settings.

[image]

NothFie, 2025-02-08:
@MartinDrost @jeffreylanters This is a unique bug of Organic Tree(Default) and we are working on it. For now I suggest you try Slim Tree as an alternative temporarily.

closed 2025-02-24 by NothFie