bambulab/BambuStudio #5865: Not possible to select TPU for AMS in printer filament setup

alexeyinthefields, 2025-01-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 12.7.2

### Additional system information

Model Name:	iMac
  Model Identifier:	iMac19,1
  Processor Name:	8-Core Intel Core i9
  Processor Speed:	3,6 GHz
  Number of Processors:	1
  Total Number of Cores:	8
  L2 Cache (per Core):	256 KB
  L3 Cache:	16 MB
  Hyper-Threading Technology:	Enabled
  Memory:	64 GB
  System Firmware Version:	2020.40.3.0.0
  OS Loader Version:	540.120.3~37
  SMC Version (system):	2.46f12

### Printer

Bambu Lab X1 Carbon Combo with AMS

### How to reproduce

1. Go to Device Tab
2. Go to Filament Selection
3. Select Generic TPU for AMS

### Actual results

In printer: 
After selection of Generic TPU for AMS material in printer Filament Tab you receive system message: TPU for AMS is not compatible with AMS)). The material is not selected.

In Bambu Studio: 
After selection of Generic TPU for AMS material in printer Filament Tab material selection disappears and the slot is not defined.

### Expected results

To be able to select Generic TPU for AMS material in printer and Bambu Studio.

### Project file & Debug log uploads

[debug_Mon_Jan_27_11_16_35_1601.log.0.zip](https://github.com/user-attachments/files/18556470/debug_Mon_Jan_27_11_16_35_1601.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

djblue81, 2025-01-30:
It is not a Bug, it is a Feature.
TPU is to soft to act in the AMS.

moe001, 2025-02-02:
This is not a bug. You can choose to skip the blacklist check in the developer mode section of the software settings, which will allow you to select TPU material. However, this will almost certainly lead to hardware failures that require disassembly and repair.

closed 2025-02-05 by zhimin-zeng-bambulab