bambulab/BambuStudio #5870: The "Check for Update" function in the Help menu does not provide the 'truth'

PierreDeQuebec, 2025-01-27:
### Bambu Studio Version

v1.9.5

### Where is the application from?

Bambu Lab github releases

### OS version

Kubuntu 24.10

### Additional system information

With Bambu Studio v1.9.5 (Appimage), the "Check for Update" function in the help menu told me there was no new version! Surprise! Github was up to version v1.10.1.

Operating System: Kubuntu 24.10
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.6.2
Kernel Version: 6.11.0-13-generic (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5700U with Radeon Graphics
Memory: 29.3 Gio of RAM
Graphics Processor: AMD Radeon Graphics

### Printer

Bambu Lap P1P

### How to reproduce

Use the fonction "Check for Update " from the help menu.

### Actual results

Message from the Bambu Studio: "This is the newest version". Which is clearly not true.

### Expected results

The message should be: "A new version is available on the Github site".

### Project file & Debug log uploads

No project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-02-01:
hello @PierreDeQuebec 

thanks for reporting this issue

we will treat it as a feature request

JohnSmithCYM, 2025-02-05:
We will check 'Check for Update' function on Ubuntu soon, thanks for your effort.