bambulab/BambuStudio #5873: Common clipboard for Bambu Studio multiple instances,

arknee, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
When I'm printing something with a lot of pieces, and I just need one, I have to modify the project, which I don't always want.

**Describe the solution you'd like**
I'd like to be able to cut/copy and paste object between instances of Bambu Studio

**Describe alternatives you've considered**
Just that, have a common clipboard for all the instances of Bambu Studio

**Additional context**
Nothing else, thanks. :)

EricDimitri, 2025-01-30:
This will be very helpful. Great if you are able to implement!

JonLaliberte, 2025-02-01:
Yeah this should really be a thing.

TechlyAccurate, 2025-02-10:
I find myself regularly having to import geometry from file after I already have the it copied to the clipboard from another instance. Definitely please add this.

grassfish1122, 2026-11-02:
we need it!

Mujin-code, 2026-11-02:
你好，如果使用仅导入零件的功能，是否可以满足需求呢 @grassfish1122

grassfish1122, 2026-11-02:
> 9\. 你好，如果使用仅导入零件的功能，是否可以满足需求呢 [@grassfish1122](https://github.com/grassfish1122)

太难用了， 我得先隔离零件，导出，再在另一个项目里导入。
这个功能很实用而且并不复杂，早在25年9月相关的issue就被提出并且分配了，到现在还没完成。

Mujin-code, 2026-11-02:
好的，感谢你的反馈，我们会评估一下这个需求