bambulab/BambuStudio #5875: Filament cutting not preceeded by cooling

pydavem, 2025-01-29:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS Ventura 13.7.2

### Printer

Bambu Lab A1

### How to reproduce

Normal slice and print

### Actual results

After attempting to cut filament during a colour change, artifacts (poop) are brough onto the print plate

### Expected results

Prior to this version the cooling fan would cool the filament befire attempting to cut

### Project file & Debug log uploads

This is with all prints I have attempted.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-02-01:
hello @pydavem 
thanks for reporting this issue

could you give some pictures on the artifacts?

pydavem, 2025-02-02:
hi @lanewei120,

I have switched off and on my printer, and now there is no issue, the filament is being cooled before each cut attempt as it was previous to the software upgrade, so I assume it was an issue with the printer and not the software.

Thanks for looking into this though, but I think this fault report can be closed as 'user error'!

lanewei120, 2025-02-06:
Hello @pydavem 

ok, we will close it
thank you for reporting this issue

closed 2025-02-06 by lanewei120