bambulab/BambuStudio #5876: Cannot open dropdown to create new printer or nozzle

brha8286, 2025-01-30:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.2 sequoia

### Additional system information

m1 macbook air, gen 1.  

The dropdowns do not open to select `Select Model` if in `Create Printer`, or `Select Printer` if in `Create Nozzle For Existing Printer`

### Printer

X1C

### How to reproduce

1. click on printer/nozzle selection
2. click on create printer
3. click on `Select Model`

### Actual results

The dropdown doesn't drop down.

[image]

### Expected results

I'd expect the drop downs to you know, drop down

### Project file & Debug log uploads

[debug_Wed_Jan_29_20_54_34_1578.log.0.zip](https://github.com/user-attachments/files/18596856/debug_Wed_Jan_29_20_54_34_1578.log.0.zip)
[tag_01.3mf.zip](https://github.com/user-attachments/files/18596855/tag_01.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

brha8286, 2025-01-30:
Actually "select model" does work after I've selected a vendor.  "Select printer" still broken though

DanBao-Bambu, 2025-02-20:
This option displays all of your visible models, but from the log, this expectation conflicts with your results. You can first select the first option and fill in the model you want to create. It has the same effect as the second option, except that using the second option will be more convenient in creating existing models. We apologize for any inconvenience caused and will continue to monitor this issue.