bambulab/BambuStudio #5879: AMS Filaments Configurations always disappear on LAN "Only"

GuiDjee, 2025-01-31:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 64bit last update

### Printer

BambuLab X1-Carbon

### How to reproduce

1. Go on "LAN Only" Mode
2. Connect Print with codes
3. Add Filaments in AMS on BambuLab scren
4. Lucky you if when you add one, another don't disappear...

### Actual results

- On LAN Mode, print are always lost connection... always have to reconnect it
- K value always deleted, I don't know why??
- When I set-up filament on the AMS draw, when I add one, antoher disappear... boring.. that mean all AMS become empty after wild...

I kown after all my filament... can I have the opportunity to introduce k factor without calibration??

### Expected results

that it works of course !!!

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-05:
Thank you for your feedback. Could you provide a video recording and log files of the operation steps? This would be helpful for us to reproduce and analyze the problem.

GuiDjee, 2025-02-07:
I don't know how to show you something by video...
I did all the automatics calibration of the Factor K.. everythings is recording in the print...
I can see the correct factor K in the page "Print" or "Appareil" in french... I use the printer, everything is ok...
After the day, I switch off everything and the day after I back on BambuLab, all the Factor K are gone... ?
When I go on "manage the factor k" in the "calibration" page, my automatics K calibration is still here... but don't wont go back in the printer...
What I do wrong ?

[image]
[image]