bambulab/BambuStudio #5881: SVG Import Resize Is Annoying

JonLaliberte, 2025-02-01:
**Is your feature request related to a problem? Please describe.**
SVG import could be a better process.

**Describe the solution you'd like**
When importing a SVG, you can only type in the size if you type both the width and height. If you want them to stay in the same ratio the only option is the weirdest UI I've ever seen - it looks like a text field, but it's actually like a slider? Why!?
Let us enter in the width or height and let us lock or not the ratio.

Haidiye00, 2025-02-01:
[image]

[image]
Hello, there are two options: one is to maintain a proportional size definition, and the other is to fully customize the length and width input. I don't fully understand the confusion you're saying. Can you take a screenshot or record a GIF to explain your confusion? Thank you~

JonLaliberte, 2025-02-01:
When proportion is locked, you can’t _type_ in the dimensions, you can only drag the slider and it adjusts the dims proportionally. 
When unlocked you can enter in both dims, but it doesn’t adjust proportionally so you have to enter in both dims.

Haidiye00, 2025-02-01:
hello，Are you saying that the slider that maintains the proportion can also be inputted？
Like this ? 
[image]

[image]

JonLaliberte, 2025-02-01:
Yes, that would be a good fix, and at least right now would solve the problem I was having. 

Ideally, there would be an option when typing in the size to set either the width or height and have it calculate the other dimension proportionally.
Kind of like how illustrator does it:

[image]

Not the link 🔗 button. When on it scales either dimension if you type in one of them. Just like the scale tool in Bambu Slicer does.

Haidiye00, 2025-02-02:
Hello, we understand your needs and will evaluate and consider whether to improve. Thank you for your suggestion~