bambulab/BambuStudio #5884: Crashes GPU

Gnoshious, 2025-02-01:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Processor	AMD Ryzen 9 5900X 12-Core Processor               3.70 GHz
Installed RAM	32.0 GB
System type	64-bit operating system, x64-based processor

Edition	Windows 10 Home
Version	22H2
Installed on	‎27/‎05/‎2021
OS build	19045.5371

Graphics
AMD Radeon RX 6800 XT

### Printer

Bambu A1

### How to reproduce

1. open new project
2. add object
3. select and move object with multiple objects on plate. 

### Actual results

My screens go blank and can't access PC. Have to restart PC.

### Expected results

I should be able to reposition the objects on the plate. To set up a job run.

### Project file & Debug log uploads

I am usinf stl file and dont get to save the project, So I can't give you this because it keeps crashing.

It did this on another project and I had to print 1 stl file at a time. But that is not efficeint.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Gnoshious, 2025-02-06:
> Thanks for your feedback, and can you offer more information for us like:
>
>    - Dmp file, which is under log folder
>    - Are you ok with previous version (like V1.9.7.52) on your computer?

Hi

Where exactly it the damp file? I'm battling to find it.

This is the first version of Bambu studio in that I'm using. I'm a new customer.

Thanks
Manuel

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

Haidiye00, 2025-11-19:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.
If there are any further questions, let's continue the discussion. Thank you