bambulab/BambuStudio #5887: Base pattern not generating for organic supports

Wysoc0, 2025-02-02:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

GPU: Radeon 780M

### Printer

Bambu Lab A1 Mini

### How to reproduce

Support:
Type = tree(auto)
Style = Organic
Advanced:
Base pattern = Honeycomb

### Actual results

[image]

### Expected results

This was available and working in Bambu Studio 1.09.07.52

[image]

### Project file & Debug log uploads

[Bambu Studio 1.10 issue.3mf.zip](https://github.com/user-attachments/files/18632947/Bambu.Studio.1.10.issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NothFie, 2025-02-05:
@Wysoc0 Thanks for your report. We are looking into this issue.
And may I ask if the reason you need this infills (which may be useless in most situations) is because the current slicing result has some floating support areas? If yes, then this is another bug that we are fixing.

Wysoc0, 2025-02-05:
> [@Wysoc0](https://github.com/Wysoc0) Thanks for your report. We are looking into this issue. And may I ask if the reason you need this infills (which may be useless in most situations) is because [...]

Mainly becasue, with printing with 0.2mm nozzle and 0.04-0.06 layer lines, for small but tall things (miniatures/figurines) hollow supports were breaking easily (not breaking off the build plate, but breaking further along the print - along 1/3 or 1/2 of the support). Honeycomb, or any other infill made the supports much stronger and lowered the number of fails.

tukez, 2025-02-08:
I've also had this problem for a long time, quite annoying since there are always some support failures and it is a guessing game should I continue the 10+ hours print or not (and I have to keep monitoring the printing).

ajgp, 2025-07-08:
I would like to add I am also having this issue on V2.1.0.59 on Windows & 2.1.1.52 on Linux (OpenSuse)

pbrouwers, 2025-07-09:
Issue still exists in version 2.1.1.52

Wylieuk, 2025-09-28:
I've found that changing the support base pattern only effects the support if type is set to 'Tree Hybred' I have version 2.2.2.56 - please fix, my supports often fall apart with no infill so I'm force to use this one working option.