bambulab/BambuStudio #5888: Beta version Crash very often, especially when mac goes to screensaver

laurent-qantum, 2025-02-02:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15.1

### Additional system information

Apple M1 Max
64Gb RAM
4Tb SSD

### Printer

Bambulab X1C

### How to reproduce

Start the application, let it run
Put the mac on hold using screensaver
Wait something like 20 minutes
Come back : the application has disappeared 

### Actual results

The application crashed

### Expected results

The application must stay alive

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/18633019/Archive.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-1, 2025-02-05:
Thank you for your feedback. Could you please provide the crash and DMP log files?

BoBoYu-1, 2025-02-06:
Could you please provide us with the Crash and DMP log files for us to analyze? We haven't been able to reproduce this issue here. Thank you.The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

laurent-qantum, 2025-02-07:
Well it has already been added in my initial post. The Archive.zip contains the logs...

laurent-qantum, 2025-02-08:
Okay here are two most recent

[IPS-BambuStudio.zip](https://github.com/user-attachments/files/18718140/IPS-BambuStudio.zip)

 ips files from the DiagnosticReports folder.
For your information it constantly crashes. I have started a print this night, the I get to bed, wake up and bambustudio wasn't running anymore. It always happens...

TinyMito, 2025-02-11:
Hello, I would like to express my agreement with this post and upvote it. The M1 Pro with macOS 15.3.0 and 15.3.1 continues to experience crashes with Bambu Studio when the screensaver is activated.

Here are all the IPS crash log from macOS spread over several days.

[IPS-Archive.zip](https://github.com/user-attachments/files/18742335/IPS-Archive.zip)

JohnSmithCYM, 2025-02-11:
We have just launched new plugin for Win/Mac/Linux, please restart Studio(2 times better, to make sure the plugin have updated) and see if the crush continue.
Thanks for all the feedback.

laurent-qantum, 2025-02-11:
From my side things are getting worse. BambuStudio don't want to start anymore. The application seems to be alive but no window is visible :/

BoBoYu-1, 2025-02-11:
Could you provide the new log file? Then clean up the environment and reinstall the studio to see if it works; we haven't been able to reproduce the issue here.

laurent-qantum, 2025-02-11:
Tried to reinstall the official version (instead of the beta version which definitely doesnt want to show any window).
The official version shows windows, and crashed when launching a print.
Attached is the crash log.

[bambu_crash_log.txt](https://github.com/user-attachments/files/18751332/bambu_crash_log.txt)

laurent-qantum, 2025-02-11:
Here it is.
You will that now there's also a lot of debug_Network logs...
Things are going terrible right now because I cannot print anymore, neither with the official or beta version !! Please do something...

[Bambu-Logs.zip](https://github.com/user-attachments/files/18754628/Bambu-Logs.zip)

BoBoYu-1, 2025-02-12:
Hello，Did you perform any special operations before the crash occurred in the first Beta version? Was it just that the application crashed immediately after unlocking the screen and opening it? Or was there any special setting in your environment? Please tell us in detail. We would like to simulate your environment to reproduce this issue.

JohnSmithCYM, 2025-02-13:
@laurent-qantum Can you offer your printer type and firmware version?

laurent-qantum, 2025-02-13:
> Hello，Did you perform any special operations before the crash occurred in the first Beta version? Was it just that the application crashed immediately after unlocking the screen and opening it? Or [...]

It depends on which crash we're talking about. The one that occurs when on screensaver, nothing special. Sometimes I launched a print before the computer goes to screensaver, sometimes I just started BambuStudio, loaded a model and that's all.
For the crash that make the main window disappear, I have loaded a model of mine with different pieces, duplicate one and it crashes when I tried to move it.

JohnSmithCYM, 2025-02-13:
@laurent-qantum Thanks for your feedback, 
- Crash about screensaver: Do new Studio Beta version work well now?
- Crash about mainwindow disappear: Can this question reproduce now ?And if you can, offer a 3mf file to us, if you can record your operations and send here would be better.