bambulab/BambuStudio #5890: 文字添加不支持中文

G-hangogh, 2025-02-03:
### Bambu Studio Version

Bambu_Studio_mac-v01.10.02.64和Bambu_Studio_mac-v01.10.01.50都不行

### Where is the application from?

Bambu Lab Official website

### OS version

mos15.2

### Printer

P1S

### How to reproduce

文字无法输入中文到物体上，选任何字体都不行，只有开始的一下可以显示其中一些文字，只要更改文字，就再也不显示中文了

### Actual results

[image]

### Expected results

支持中文输入

### Project file & Debug log uploads

[立方体.3mf.zip](https://github.com/user-attachments/files/18636306/3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-05:
Hello, can you provide this font? Thank you

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If you have any further questions, we will analyze again

jianli-ninebot, 2026-03-15:
I'm on Mac Bamboo studio 2.5.0.66, the latest version. But I stilll can not input Chinese Characters. Which beta release can solve this?