bambulab/BambuStudio #5894: Lan only mode does not work

xoores, 2025-02-03:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Gentoo

### Additional system information

Bambu Studio cannot connect to the printer in the LAN only mode. The "add printer" dialog just closes with no error message, so I tried to do a packet capture and I can see that printer returns an error JSON:

```json
{"login":{"command":"login_report","reason":"{\"devmsg\":\"region or env unmatch\",\"err_code\":83968056}","sequence_id":1000,"session_id":-1,"status":"FAILURE"}}
```

Full sequence:

[image]

My computer is in the same LAN (although each in different subnet) and traffic can go between these two (as shown in the packet capture).

### Printer

Bambu X1C

### How to reproduce

1. Try to add printer in Lan Only mode in Bambu Studio
2. Fill out access code, IP and serial number
3. click add

### Actual results

Nothing. Printer is not added and no error is displayed, leaving me wondering wtf 😅 

### Expected results

Added printer in Bambu studio and ability to print in LAN only mode.

### Project file & Debug log uploads

No log files...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-02-06:
The same with #5833

JohnSmithCYM, 2025-02-06:
@xoores You can try [Beta version](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64), we have tried to fix it in the new network plugin.

walterwongbbl, 2025-02-06:
@xoores 
Hi.  Please check again if you have switched to LAN mode, as the error message indicates that the environment or region is incorrect. This message typically appears when binding in non-LAN mode.
{\"devmsg\":\"region or env unmatch\"

xoores, 2025-02-06:
> [@xoores](https://github.com/xoores) You can try [Beta version](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64), we have tried to fix it in the new network plugin.

@JohnSmithCYM: Tried version `01.10.02.64` but unfortunately error is still the same 🙁 After putting in the IP and access code, Bambu Studio tries to contact the printer and gets rejected...

Relevant communication snippet from wireshark:
[image]

@walterwongbbl Yes it is. Mobile app does not work also, it is telling me the printer is in LAN Only mode. 🤷

lone0, 2025-02-23:
I made an attempt with version 01.10.02.72 and got the following:

..:.{"login":{"command":"detect","sequence_id":"50006"}}......{"login":{"sequence_id":1000,"command":"login_report","status":"FAILURE","reason":"{\"devmsg\":\"parser config failed\",\"err_code\":83968024}"}}..

My operations: Device -> + -> Bind with Access code -> Enter IP and access code.

xoores, 2025-02-25:
@BoBoYu-1 @walterwongbbl

Is there anything new? The Lan only mode still does not work 🙁

ElectronicElephant, 2025-03-25:
I don't think this bug has been fixed in the latest release (1.10.2.76)

M1-Olsson, 2025-05-19:
I start to wonder if they ever will fix the problem with subnets and studio, they have had several "attempts" but its still not working. I wonder if they even test the fixes they do as its very simple to replicate.

Use Orca slicer, they have solved this and it works great.

xoores, 2025-05-19:
Yeah, I'm also slowly starting to think that non-functioning LAN-only mode is not a bug, but a feature 😅 

I did not know about Orca - will try that later, that might be nice alternative if it works!

xoores, 2025-05-19:
Wow, Orca Slicer actually **does pair** with my X1C and can control e.g. lamp, but does not actually print and head/plate controls don't do anything...

Probably some kind of firmware problem - but it definitely shows that the pairing cat actually work and can be fixed.

M1-Olsson, 2025-05-22:
> Wow, Orca Slicer actually **does pair** with my X1C and can control e.g. lamp, but does not actually print and head/plate controls don't do anything...
> 
> Probably some kind of firmware problem - but it definitely shows that the pairing cat actually work and can be fixed.

It seems the last version of Bambu Studio replaced some files it uses and broke it...

hegge25, 2025-07-08:
Any news on this? I am also struggling with this on my H2D at work.

cinderblockgames, 2025-07-08:
Temporary workaround until this is fixed: https://github.com/Rdiger-36/StudioBridge

hegge25, 2025-07-09:
Wow, that was easy. The printer is now connected. Can't figure out why bambu doesn't just fix this themselves. Thanks.

savrielynck, 2025-07-28:
Face with kind of the same errors on a NixOS. 
I can do everything with the printer with Orca Slicer (Send print/Camera/change AMS config/ etc...), with Bambu Studio I get free error as soon as I enter the Access Code.

I haven't dig deeper since Orca Slicer works but I would like to remove the Developer Mode from the printer since Orca fails the Auth check when sending prints. (Or if someone has a solution for the Auth check I would take it :P)

### Error log
```
free(): invalid pointer
Aborted (core dumped)
```