bambulab/BambuStudio #5895: New Bambu Lab x1 Carbon - bed temp showing 88 Deg C

c0l1n-eng, 2025-02-03:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10.0.19045 Build 19045

### Additional system information

Processor	11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz, 2496 Mhz, 8 Core(s), 16 Logical Processor(s)
64 Gb Ram

### Printer

Bambu labs Carbon with AMS

### How to reproduce

When sending any file to to print -the bed temp shows 88 deg. 
a message pops up saying High Bed temperature may lead to filament clogging in the nozzle. You may open the chamber door [1000c001 000501]
This is a brand new printer delivered Thursday last week 
We have purchased 2 printers so far but was hoping if these worked fine would purchase another 12 printers with AMS

[image]

 

### Actual results

Nothing will print from the printer. When feeling the bed it doesn't feel hot

### Expected results

I expected it to work ? 

### Project file & Debug log uploads

[00001965.zip](https://github.com/user-attachments/files/18644751/00001965.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-02-06:
Thanks for your feedback, and we can't reproduce this problem, both gcode(M140 and M190) and real print show no temp error,could you offer more information for us to analyze ?
- Printer firmware version and log
- Studio log
And if you can still reproduce this problem, please let us know

c0l1n-eng, 2025-02-06:
Good morning,

As I said we have 2 printers the same. One will print perfectly and the other will not.
The bed temp shows 90 deg temp but is cold to the feel with a hand.
Printer firmware - 01.08.02.00(latest)

This is stopping me purchasing further machines at the moment.

Many Thanks,
Colin Evans

Senior Product Engineer

***@***.******@***.***>
07803 138526

c0l1n-eng, 2025-02-06:
***@***.***

This is a screen shot after the machine has been idle all day

Many Thanks,
Colin Evans

Senior Product Engineer

***@***.******@***.***>

c0l1n-eng, 2025-02-07:
I CAN NOT use this printer at the moment. PLEASE help sort this out.

THIS IS URGENT!!

Many Thanks,
Colin Evans

Senior Product Engineer

***@***.******@***.***>