bambulab/BambuStudio #5899: PC Freeze on Slice with Arachne

AggieTutor, 2025-02-04:
### Bambu Studio Version

01.10.02.16

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Threadripper 2970
Memory: 35GB
GPU: NVIDIA 3090

### Printer

Bambu Lab P1S and X1C

### How to reproduce

1. Load a Large or complex model from Makerworld.
2. Set Wall Generator to Arachne
3. Slice.

### Actual results

1. Progress to 15%
2. PC Freeze and must be restarted with power button.

### Expected results

Progress to 100%

### Project file & Debug log uploads

Project Title: Moon Globe for LED Lamp 001
Model link: https://makerworld.com/en/models/31191#profileId-27822

Log file not available

### Checklist of files to include

- [x] Log file
- [x] Project file

JohnSmithCYM, 2025-02-07:
[image]

Thanks for your feedback and I have tried the model with my PC(win11 , i7, 32GB and RTX3060) and it did cost me 10 minutes to finish the slicing, yes, it did cost a long time but it didn't crash.

Our developer has realized most time cost was due to overhang detection and we will try to optimize it in the future.You can use Classic wall to get your slice in a short time for now, and thanks for your feedback again.

XunZhangBambu, 2025-06-03:
@AggieTutor Hi, we have optimized the algorithm in our latest verion 02.01.00.59. It now only takes about 1 minute to slice