bambulab/BambuStudio #5900: baambu studio crashes most of the time when sending print job to my a1 mini

remizca, 2025-02-04:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Processor	11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz (2.69 GHz)
Installed RAM	16.0 GB (15.7 GB usable)
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 11 Home Single Language
Version	24H2
Installed on	‎01/‎02/‎2025
OS build	26120.3073
Experience	Windows Feature Experience Pack 1000.26100.50.0

From dxdiag:
Operating System: Windows 11 Home Single Language 64-bit (10.0, Build 26120) (26100.ge_release.240331-1435)
Language: English (Regional Setting: English)
System Manufacturer: ASUSTeK COMPUTER INC.
System Model: ASUS TUF Gaming F15 FX506HC_FX506HC
BIOS: FX506HC.315 (type: UEFI)
Processor: 11th Gen Intel(R) Core(TM) i5-11400H @ 2.70GHz (12 CPUs), ~2.7GHz
Memory: 16384MB RAM
 Available OS Memory: 16122MB RAM

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Slice Plate.
2. Print Plate.
3. flow dynamics calibration (either on or off) nd timelaapse off.
4. hit send.
5. progress gets stuck at 30% and then crashes.

### Actual results

progress gets stuck at 30% and then software crashes.

### Expected results

Printer receives job and proceeds to print.

### Project file & Debug log uploads

[crash_Tue_Feb_04_15_04_34_0_done.log](https://github.com/user-attachments/files/18653053/crash_Tue_Feb_04_15_04_34_0_done.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

JohnSmithCYM, 2025-02-06:
Thanks for your feedback and we have checked A1 mini with new studio for like 5 times and no crash happened. Can you reproduce this problem now? What's more, the log file you offered is not enough for us to analyze, can you package the hole dmp file (or zip the hole log folder) then send to us?

remizca, 2025-02-06:
> Thanks for your feedback and we have checked A1 mini with new studio for like 5 times and no crash happened. Can you reproduce this problem now? What's more, the log file you offered is not enough for us to analyze, can you package the hole dmp file (or zip the hole log folder) then send to us?

I'll try and check if I encounter it again when I get home, I am currently at work. But what I can say is that it does not always happen but when it does it occurs consecutively (crashes and occurs again when trying to send the print job a few more times) until it decides to just send the print job successfully.

Yes, alright. I'll zip the whole log folder and send it to you guys when I get home.

remizca, 2025-02-06:
[log.zip](https://github.com/user-attachments/files/18689549/log.zip)

[1__transport_rover_ByPlate.zip](https://github.com/user-attachments/files/18689551/1__transport_rover_ByPlate.zip)

happened again trying to print this one. i zipped the whole log folder and lso attached the zipped 3mf file.

remizca, 2025-02-08:
any update on this? it's frustrating when it happens. Multiple bambu studio crashes until it decides to pity me and send the job to the printer.

walterwongbbl, 2025-02-10:
Hi.  @remizca 
I noticed from the logs you provided that your network plugin version is not the latest. Could you please delete your current network plugin and download it again?  
Or reinstall Bambu Studio, but make sure to delete its configuration file directory.

remizca, 2025-02-10:
> Hi. [@remizca](https://github.com/remizca) I noticed from the logs you provided that your network plugin version is not the latest. Could you please delete your current network plugin and download it again? Or reinstall Bambu Studio, but make sure to delete its configuration file directory.

1. how do i delete the network plugin?
or is it easier to just reinstall bambu studio? i currently have the beta.
2. will my filament profiles remain untouched if i uninstall bambu studio and then reinstall?
3. i also have orcaslicer installed on the same machine, should i uninstall that too to ensure the network plugin is removed?

ah, also. i'll probably report back again later because i'm still at work. i'll do the instructions later when i'm done with work.

JohnSmithCYM, 2025-02-10:
@remizca 
1. On windows: C:\Users\UserName\AppData\Roaming\BambuStudioBeta\plugins, and delete all files under that(they are all plugin dlls)
2. You can log in your account to make sure your profile had upload to cloud first then each time you log in you can pull your profile from cloud.
3. You can get a try, and is that orca slicer worked well with your a1 mini?

remizca, 2025-02-10:
> [@remizca](https://github.com/remizca)
>
>     1. On windows: C:\Users\UserName\AppData\Roaming\BambuStudioBeta\plugins, and delete all files under that(they are all plugin dlls)
>
>     2. You can log in [...]

So I tried that one but also uninstalled orca and bambu studio to make sure everything's clean except the user data (the profiles and stuff).
So far it's going good, i'll have to observe more if it happens again.
also, yeah orca worked great but I'm not using orca that much anymore(i was using it back when i'm using klipper) except for using the calibration tools that aren't available in bambu studio (mainly the temp tower calibration).

remizca, 2025-02-10:
Project File: https://pixeldrain.com/u/XxPu2iab
Log Folder: https://pixeldrain.com/u/7HX3U6TL

happened again.
are the dmp files supposed to be that large?

remizca, 2025-02-15:
up

remizca, 2025-02-20:
so i downloaded the new update and installed a fresh one. still happening. is my system the damn problem here??? i also found out at some forum that setting the affinity of the software via task manager to half of the cores somehow fixes it and so far i haven't got any crashes when i did that but it's so troublesome to do some extra steps every time i launch BS. what the hell is happening?

remizca, 2025-04-05:
I'm on v2.0.1.50 and this software still keeps crashing when sending print job to printer.

XinZhangBBL, 2025-04-23:
@remizca Thank you for your feedback.  We have found the problem on your network plugin version. It's solved on latest network plugins, you could try to update  it

XinZhangBBL, 2025-04-25:
_im getting tired of this man. can't even print anything right now._ 

@remizca Please tell us what happend. The program keep crashing or not ? 
Or, If you can not send print job. Please try the fixed version 
https://github.com/bambulab/BambuStudio/releases
[image]

remizca, 2025-04-25:
> _im getting tired of this man. can't even print anything right now._
> 
> [@remizca](https://github.com/remizca) Please tell us what happend, maybe provide some vedio or images. The program keep crashing or not ? Or, If you can not send print job. Please try the fixed version https://github.com/bambulab/BambuStudio/releases [image]

okay. currently at work but i'll reply anyway.
so last night after the update (the one that popped up), i navigated to makerlab via BS. created a flexi there and then downloaded it, opened the 3mf with BS. sliced okay. tried sending the print job. loading got stuck at 30% and then just crashed. repeated countless times i just didnt count anymore, just ran BS again when it crashed hoping one sending pushes through. it finally did send the print job at some point though after a lot of crashes.

XinZhangBBL, 2025-04-25:
okay. currently at work but i'll reply anyway.
@remizca Thank you for your efforts. the problem has been solved on new version?  If crashed, please contact us, and attach dmp file and log files. we will look in the issues as soon as possible.

remizca, 2025-04-25:
> okay. currently at work but i'll reply anyway. [@remizca](https://github.com/remizca) Thank you for your efforts. the problem has been solved on new version? If crashed, please contact us, and attach dmp file and log files. we will look in the issues as soon as possible.

haven't printed anything yet after the flexi, will report back if i experience the same issue. thank you

remizca, 2025-04-26:
@XinZhangBBL 

update: installed the one mentioned above and i tried a multicolor test with a benchy.

[Log Files](https://pixeldrain.com/u/1xRJeBmR)

[3DBenchy.zip](https://github.com/user-attachments/files/19921743/3DBenchy.zip)

XinZhangBBL, 2025-04-27:
@remizca 
From the attached log files, I recognized a crash. The crash issue wille be solved on next release of network plugins.
[image]

And which steps stop you from sending print, can you attach a screenshot?

> [@XinZhangBBL](https://github.com/XinZhangBBL)
> 
> update: installed the one mentioned above and i tried a multicolor test with a benchy.
> 
> [Log Files](https://pixeldrain.com/u/1xRJeBmR)
> 
> [3DBenchy.zip](https://github.com/user-attachments/files/19921743/3DBenchy.zip)
> 
> small update: for some reason only sending the file to the printer storage works, so far tested only 2 files and it did not crash. after sending i'd have to start the print directly from the printer screen.

remizca, 2025-04-27:
> [@remizca](https://github.com/remizca) From the attached log files, I recognized a crash. The crash issue will be solved on next release of network plugins. [image]
> 
> And which steps stop you from sending print, can you attach a screenshot?
> 
> > [@XinZhangBBL](https://github.com/XinZhangBBL)
> > update: installed the one mentioned above and i tried a multicolor test with a benchy.
> > [Log Files](https://pixeldrain.com/u/1xRJeBmR)
> > [3DBenchy.zip](https://github.com/user-attachments/files/19921743/3DBenchy.zip)
> > small update: for some reason only sending the file to the printer storage works, so far tested only 2 files and it did not crash. after sending i'd have to start the print directly from the printer screen.

ohh nice, hopefully that fixes it for me.

so before BS crashes, when i click print plate or print all, and then send the print, the progress bar goes up to 30%, hangs and then crashes.

but when i only do send or send all, it sends it successfully to the printer. and then i either go to the device tab->storage->then print from there or use the A1 mini screen to print the file and somehow it prints it now because the file is already at the storage of the printer.

XinZhangBBL, 2025-04-27:
We will inform you as soon as we release the fixed network plugins.
How to walk around the problem: you can  try set the dns server to 8.8.8.8, maybe helpful

> ohh nice, hopefully that fixes it for me.
>
> so before BS crashes, when i click print plate or print all, and then send the print, the progress bar goes up to 30%, hangs and then crashes.
>
> but when i [...]

remizca, 2025-05-19:
@XinZhangBBL hello. so far no crashes yet with direct sending print (choosing Print Plate option). but recently i just straight up use Send option instead too just so i don't wonder if BS will crash or not. im just doing a few more extra steps but at least im not thinking if BS will crash.

XinZhangBBL, 2025-05-20:
@remizca Thank you for your feedback and this issue will be closed. 
When you encounter any problems, feel free to provide feedback to us.

closed 2025-05-20 by XinZhangBBL