bambulab/BambuStudio #5902: No connection to Server error message

palbiez, 2025-02-04:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 Mini

### How to reproduce

Not reproduceable. 
From one time to another the error message No connection to server was shown in Bambu Studio. 
The debug log shows: 
`2025-02-04 20:33:20.240654[Thread 0x000013d4]:Slic3r::NetworkAgent::send_message error: network_agent=0000025098AEC740, ret=-2, dev_id=0309DA4A1502044, json_str={"pushing":{"command`

### Actual results

No connection to server

### Expected results

Server should is reachable

### Project file & Debug log uploads

[debug_Tue_Feb_04_19_29_04_23188.log.zip](https://github.com/user-attachments/files/18662509/debug_Tue_Feb_04_19_29_04_23188.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

palbiez, 2025-02-04:
storage network test fails also

palbiez, 2025-02-05:
This morning it is working well again.
Checked the logs from yesterday and compared with today:
Yesterday:
`2025-02-04 21:54:49.980883[Thread 0x00003e84]:new version = 01.10.02.64, skip version = 01.10.02.64
2025-02-04 21:54:50.041671[Thread 0x00003e84]:keep alive = 0, push_count = 0, count = 1
2025-02-04 21:54:50.041671[Thread 0x00003e84]:Slic3r::NetworkAgent::send_message error: network_agent=00000248AD981960, ret=-2, dev_id=0309DA4A1502044, json_str={"pushing":{"command":"pushall","push_target":1,"sequence_id":"20000","version":1}}, qos=0
2025-02-04 21:54:50.041671[Thread 0x00003e84]:command_get_version
2025-02-04 21:54:50.041671[Thread 0x00003e84]:Slic3r::NetworkAgent::send_message error: network_agent=00000248AD981960, ret=-2, dev_id=0309DA4A1502044, json_str={"info":{"command":"get_version","sequence_id":"20001"}}, qos=1
2025-02-04 21:54:50.044665[Thread 0x00003e84]:MediaPlayCtrl switch machine: 0309DA4A1502044
2025-02-04 21:54:50.044665[Thread 0x00003e84]:MediaPlayCtrl::SetStatus:  tutk_state: 
2025-02-04 21:54:50.053662[Thread 0x00003e84]:monitor: show_status = 24
2025-02-04 21:54:50.084916[Thread 0x00003e84]:Slic3r::GUI::Sidebar::load_ams_list: 0 items
2025-02-04 21:54:50.120423[Thread 0x0000425c]:HMSQuery: HMS info is the latest version
2025-02-04 21:54:51.038371[Thread 0x00003e84]:keep alive = 996, push_count = 0, count = 2
2025-02-04 21:54:52.038593[Thread 0x00003e84]:keep alive = 999, push_count = 0, count = 3
2025-02-04 21:54:53.036083[Thread 0x00003e84]:keep alive = 997, push_count = 0, count = 4
2025-02-04 21:54:54.040640[Thread 0x00003e84]:keep alive = 1005, push_count = 0, count = 5
2025-02-04 21:54:54.040640[Thread 0x00003e84]:Slic3r::NetworkAgent::send_message error: network_agent=00000248AD981960, ret=-2, dev_id=0309DA4A1502044, json_str={"pushing":{"command":"pushall","push_target":1,"sequence_id":"20002","version":1}}, qos=0
2025-02-04 21:54:55.040432[Thread 0x00003e84]:keep alive = 999, push_count = 0, count = 6
2025-02-04 21:54:56.040259[Thread 0x00003e84]:keep alive = 1000, push_count = 0, count = 7
2025-02-04 21:54:57.040295[Thread 0x00003e84]:keep alive = 999, push_count = 0, count = 8
2025-02-04 21:54:58.037334[Thread 0x00003e84]:keep alive = 996, push_count = 0, count = 9
2025-02-04 21:54:58.037334[Thread 0x00003e84]:Slic3r::NetworkAgent::send_message error: network_agent=00000248AD981960, ret=-2, dev_id=0309DA4A1502044, json_str={"pushing":{"command":"pushall","push_target":1,"sequence_id":"20003","version":1}}, qos=0
2025-02-04 21:54:59.048811[Thread 0x00003e84]:keep alive = 1011, push_count = 0, count = 10
2025-02-04 21:55:00.040223[Thread 0x00003e84]:keep alive = 990, push_count = 0, count = 11
2025-02-04 21:55:00.811531[Thread 0x00003e84]:Slic3r::GUI::GUI_App::persist_window_geometry::<lambda_70927d4009f80d82ffdfe06309b8c093>::operator (): received wxEVT_CLOSE_WINDOW, trigger save for window_mainframe
`
This morning
`2025-02-05 08:41:45.269062[Thread 0x00005980]:new version = 01.10.02.64, skip version = 01.10.02.64
2025-02-05 08:41:46.669385[Thread 0x00005980]:command_get_version
2025-02-05 08:41:46.670385[Thread 0x00005980]:command_get_access_code
2025-02-05 08:41:46.670691[Thread 0x00005980]:Slic3r::GUI::Sidebar::load_ams_list: 1 items
2025-02-05 08:41:46.684719[Thread 0x00005980]:command_get_version
2025-02-05 08:41:46.684719[Thread 0x00005980]:command_get_access_code
2025-02-05 08:41:46.684719[Thread 0x00005980]:Slic3r::GUI::Sidebar::load_ams_list: 1 items
2025-02-05 08:41:47.316881[Thread 0x00005980]:Slic3r::MachineObject::update_printer_preset_name 5322start update preset_name
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::PresetBundle::get_printer_names_by_printer_type_and_nozzle 1855 printer name: Bambu Lab A1 mini 0.4 nozzle
2025-02-05 08:41:47.318884[Thread 0x00005980]:The values of tray_info_idx and tray_type do not match tray_info_idx  tray_type  system_type
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 0 id GFG99 color 735DF9FF
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 1 id GFA05 color FFF144FF
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 2 id GFL03 color F72323FF
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 3 id GFL03 color 161616FF
2025-02-05 08:41:47.318884[Thread 0x00005980]:Slic3r::GUI::Sidebar::load_ams_list: 5 items
2025-02-05 08:41:47.345881[Thread 0x00005980]:command_get_version`

walterwongbbl, 2025-02-06:
@palbiez 
Hello, I checked your logs and found that it has been unable to connect to our server. Could you please try using a different network or use your mobile hotspot and try again?

[image]

walterwongbbl, 2025-02-08:
@palbiez 
We will no longer display this pop-up frequently. It will only appear when an important error occurs.