bambulab/BambuStudio #5905: Poor Mac M-chip GPU Utilization

onyxcinema, 2025-02-05:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.3

### Additional system information

M4 Pro 16 inch with 24gb of ram stutters the same amount during complicated sliced project Preview as an M1 standard. According to Activity app, it seems like the majority of the load of changing the view in the preview tab is placed on the CPU.  

### Printer

X1C, A1

### How to reproduce

1) On a Apple Silicon Mac, open Bambu Studio
2) Add a complex multicolor model and duplicate to fill bed.
3) Slice
4) Change the view of the Preview tab by left clicking and dragging.

### Actual results

Super choppy refresh of preview tab.

### Expected results

Refresh rate of the preview tab should be commensurate with the GPU processing power of the Mac.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-06:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

And can you provide 3mf？

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

onyxcinema, 2025-03-26:
Sorry I never replied with the gif. I just downloaded v2 and it seems the problem has spread to the Prepare tab as well, but it's only choppy in the viewing panel and not the rest of the UI.

[https://imgur.com/a/I34Avef](url)

And in the preview tab, once sliced, it's still super choppy to manipulate the view as you can see in the second gif.

This happens on pretty much all complex projects where a PC would do fine, but here is the project this example is based on
[https://makerworld.com/en/models/971144-light-fury-toothless-with-a-knitted-design?from=search#profileId-942811](url)

onyxcinema, 2025-11-16:
Hi I just wanted to let you know this problem persists on version 2.3.1.51