bambulab/BambuStudio #5909: 贴合紧密的多个同色模型在切片后出现预期之外的缝隙和瑕疵

moe001, 2025-02-06:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

无关

### Additional system information

无关

### Printer

p1s/无关

### How to reproduce

1.打开附件模型
2.进行切片

### Actual results

在走线预览界面，可以看到模型顶面出现了不该有的孔洞和裂隙。

虽然我不太确定他们是否有网格或浮点数的微小差异，但这种视觉贴合紧密的模型，在默认精度参数下，不应该出现这种比原缝隙还大的走线瑕疵。

### Expected results

生成平滑的模型顶面，或者每个子模型有独立的墙，不出现这种似是而非的行为。

### Project file & Debug log uploads

[翅膀1.zip](https://github.com/user-attachments/files/18681196/1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-02-06:
The same performance on V1.9.7.52.

QingZhangBambu, 2025-02-07:
@moe001 模型被提示有自相交的情况出现。表示模型中可能含有错误的点。会导致切片错误和间隙。建议先修正模型后再进行切片

[image]

moe001, 2025-02-07:
在我的版本上没有看到这个警告，是需要更加新的版本吗。另外，这个自相交模型是在软件内导入svg并进行一些模型操作后产生的，而不是来自于外部，所以这里应该还有另一个bug，我会新开一个issue来说明。

QingZhangBambu, 2025-02-07:
我使用了目前公开的最新版本。使用你的文件之后出现了错误提示。要先去除自相交对切片的影响再解决剩余部分。我怀疑还有两个部分模型轮廓不完全贴合的问题，这个在svg转模型中经常发生。

[image]

QingZhangBambu, 2025-02-07:
如果你把象脚设置成0  分辨率设置成0. 这个间隙问题会更明显

moe001, 2025-02-07:
我不太确定是否有某些设置影响了警告信息的弹出，我在相同的版本上点击切片只会弹出切片完成，如图

[image]

具体到模型相关，这里确实还有多方面的问题，我似乎遇到了多个bug，正在尝试整理

QingZhangBambu, 2025-02-07:
嗯。svg转模型这块确实有很多问题。不着急，慢慢梳理。可以把你遇到的发上来。