bambulab/BambuStudio #5911: Bambustudio crashes with new CPU

Holydroller, 2025-02-06:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

CPU: Intel Core i7-14700K
RAM: 32 GB DDR5 7200
Mainboard: Gigabyte B760M DS3H AX
SSD: Samsung SSD 990 Pro

### Printer

Bambu Lab X1C

### How to reproduce

When i Click on "Slice" the PC Crashes and restart without warnings.

### Actual results

The PC restarts.

### Expected results

When i activate only 8 CPU Cores (my CPU has 20) then everything is ok.

### Project file & Debug log uploads

No File.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-06:
Has this problem occurred before in the previous version?

ChrisCoxArt, 2025-02-16:
If the entire system restarts or shuts down, that indicates a power supply problem to some part of the system.
That could be the RAM, the CPU, or more likely your video card/GPU.
But application software cannot cause the system to shutdown or restart - only a hardware fault or the BIOS can do that.

aegwyng, 2025-05-09:
I'm having the same problem with BambuStudio. 
CPU: Intel Core Ultra 7
GPU: RTX 5080
48GB Ram (2 Slots)
Windows 11 Pro

The pc restarts even when BambuStudio is the only thing running.