bambulab/BambuStudio #5914: 导入svg生成的模型在简单分割后产生意外的非流形边并影响切片加工效果

moe001, 2025-02-07:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

无关

### Additional system information

无关

### Printer

p1s/无关

### How to reproduce

1. 打开软件，导入附带的svg
2. 点击预览窗口上侧工具栏的“剪切 [C]”
3. 在高度栏输入0.2，勾选“切割为零件”，点击执行切割
4. 再次点击上述剪切按钮
5. 在高度栏输入9.8，点击执行切割

### Actual results

程序警告切割产生了非流形边

### Expected results

对于这种svg导入产生的简单模型，不应该在切割时产生非流形边。

一些额外信息：
- 似乎切割的位置也会影响这个bug的触发，如果两次切割的高度改为5.0和8.0，则不会出现非流形边的警告。
- 无论警告后是否选择修复，似乎都会对模型的外观产生影响，比如上述0.2+9.8的模型，切片预览可见顶面出现瑕疵，这些瑕疵在切割前是没有的，并且调整分辨率等效果不明显。
- 较薄的切割厚度似乎容易出问题，但这个是不可避免的，我经常会把模型表面切几片层高厚度（如0.2）的薄片，然后修改它们的颜色，来实现多层叠加产生混色效果，如第一层黄第二层蓝来模拟绿色，这使用其他方式来实现比较麻烦。

### Project file & Debug log uploads

[翅膀1svg.zip](https://github.com/user-attachments/files/18704941/1svg.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-08:
[image]
Hello, during our research and development process, we found that cutting may inevitably produce non popular edges, but repairing non popular edges will not affect the shape of the original model.

Can you take a screenshot to illustrate the defect you mentioned? Thank you~

moe001, 2025-02-08:
瑕疵出现在切片之后，可以看到表面有奇怪的缝隙，而切片原始模型是没有这个问题的。这可能是因为一些自相交，又或者修复影响了点坐标，从而导致多个模型相交导致的。也可能与分辨率等设置有关，但我不太了解软件的底层逻辑所以只是猜测。

并且，这个情况只出现在需要修复的分割操作后，比如把0.2+9.8的分割，改成9.8+0.2，也就是先在9.8分割，再在0.2分割，那么切片后的顶层依旧是完美的（但此时最底层便会有缝隙）

个人认为这个bug让svg的分层分色变得充满不确定性，应该有某种机制去避免。我当前的办法是复制三份全部模型，并且分别设置成0.2/9.6/0.2的层高并堆叠放置，但非常繁琐。

Haidiye00, 2025-02-10:
Okay, we will conduct an analysis~