bambulab/BambuStudio #5916: 导入svg生成的模型在删除数个子模型后被软件拒绝切片

moe001, 2025-02-07:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

无关

### Additional system information

无关

### Printer

p1s/无关

### How to reproduce

1. 打开软件，导入附带的svg
2. 点击切片，可见切片正常
3. 回到“准备”标签页，在对象列表中依次右键删除“翅膀1_0”和“翅膀1_1”
4. 点击切片

### Actual results

程序拒绝切片并提示错误，但以当前状态保存工程后再打开则恢复正常

### Expected results

正确进行切片

### Project file & Debug log uploads

[翅膀1svg.zip](https://github.com/user-attachments/files/18704941/1svg.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-08:
Hello, we have identified and resolved the issue. Thank you for your feedback.

[image]
At present, you can reset the consumable wire number, and an error will exit. I'm very sorry~

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00