bambulab/BambuStudio #5919: BambuStudio 1.10.1.50 Text shape bux

PLSomebody, 2025-02-07:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.3 (24D60)

### Additional system information

MacBook Pro 16 M2 Max

### Printer

X1 Carbon

### How to reproduce

1. Try to add Text shape object
2. Try to rotate it using slider
3. Try to enter arbitrary angle using text field

### Actual results

Angle is changing when using slider
Angle is not changing when entering arbitrary angle in text field despite, that field value is remembered

### Expected results

Both methods of setting angle should work

### Project file & Debug log uploads

Any project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-08:
[image]
Hello, is this the input? I have tested it and it is normal.

Can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If you have any further questions, we will analyze again

closed 2025-07-17 by Haidiye00