bambulab/BambuStudio #5923: SVG file only partially imported

seanbo, 2025-02-07:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

I have an SVG of a logo. The text in the SVG is not being imported into Bambu Studio. See attached screenshots.

SVG Screenshot

[image]

Bambu Studio Build Plate after Import

[image]

### Printer

Bambu lab A1 Combo

### How to reproduce

1) Go to File->Import-> Import 3MF/STL/STEP/SVG/OBJ/AMF...
2) Select SVG file

### Actual results

The bounding box, the snake body and the snake eye are all imported.  The text is not.

### Expected results

The bounding box, the snake body and the snake eye and text are all imported. I have included the .svg and the .3mf file I tried to import it into.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18715511/log.zip)

[Python Palace Logo.zip](https://github.com/user-attachments/files/18715513/Python.Palace.Logo.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-02-08:
[image]

Hello, you can use Figma website or software to convert text into a path, and everything will be fine now.
https://www.figma.com/

[image]

[image]
The software does not support formats such as text or image, but it does support path very well. Thank you for your understanding~
[image]

seanbo, 2025-02-08:
Thank you.  I was able to convert the text to path in Inkscape also.  I appreciate your assistance.

closed 2025-02-08 by seanbo