bambulab/BambuStudio #5927: Exporting single plate mistakenly exports all metadata previews

spuder, 2025-02-09:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OSX 15.0

### Printer

P1S

### How to reproduce

1. Create a 3mf file with multiple plates

[image]

2. Slice and export a single plate to `foobar.3mf`

[image]

3. Unzip `foobar.3mf`, observe that `Metadata/plate_1.png` and `Metadata/plate_2.png` both exist

[image]

### Actual results

Preview metadata images are generated for _all_ plates, not just the generated plates. 

[image]

### Expected results

The slicer should only export the plate preview images for the exported plate (e.g. `Metadata/plate_1.png`

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/user-attachments/files/18723100/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

spuder, 2025-02-09:
Bug likely lies in this file. 

https://github.com/bambulab/BambuStudio/blob/master/src/libslic3r/Format/bbs_3mf.cpp#L5723

spuder, 2025-02-09:
Draft PR: https://github.com/bambulab/BambuStudio/pull/5928

Currently untested. Feedback welcome.

zhimin-zeng-bambulab, 2025-02-17:
@spuder  Thanks for reporting this issue and for providing the corresponding PR! We will refer to your PR to make corresponding changes.

spuder, 2025-08-30:
1