bambulab/BambuStudio #5929: Print head stopped.  head appears to have just stopped while printing.  Handy reported print as "cancelled".  Head not homed.

oodlexaps, 2025-02-09:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.10

### Additional system information

Intel(R) Pentium(R) Gold G5400T CPU @ 3.10GHz

### Printer

P1S with AMS

### How to reproduce

It is random.  This is the 2nd print to just get "cancelled" with no warning.

### Actual results

Usually it prints fine.  A couple times it has stopped for no apparent reason.

### Expected results

Completes the print.

### Project file & Debug log uploads

[debug_Sat_Feb_08_14_49_40_234266.log.0.zip](https://github.com/user-attachments/files/18723698/debug_Sat_Feb_08_14_49_40_234266.log.0.zip)

[binfinity-drawer-framework_115w_2x3.3mf.zip](https://github.com/user-attachments/files/18723699/binfinity-drawer-framework_115w_2x3.3mf.zip)

Print time started at 17:01, "cancelled" at 20:03

### Checklist of files to include

- [x] Log file
- [x] Project file

gaolegao-lx, 2025-03-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

oodlexaps, 2025-03-24:
This is from over a month ago.  Would I still have logs from then?
My app does not look like the above picture.
I went to Me > Support Center > Ticket History but I don't see a ticket related to this.

closed 2025-05-07 by BoBoYu-BamBu