bambulab/BambuStudio #5931: Blank home screen when opening bambu studio

camiloj04, 2025-02-10:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1s

### How to reproduce

reinstalled bambu studio. opening the application does not show home page. unable to sign in

### Actual results

[image]

### Expected results

show homepage

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18727385/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

ANicksn, 2025-02-10:
In addition, some filament profile settings windows are displayed empty in version v01.10.02.64. 

[image]

realzorro, 2025-02-17:
We are aware of this matter and we apologize for any inconvenience caused. Please try checking if the Edge browser on your system is working properly and try reinstalling BambuStudio. Please continue to follow the release of the new version.

WilliamBurkert, 2025-03-22:
Over thirty days with no resolution, is there update with what is wrong and when we can expect a fix?