bambulab/BambuStudio #5935: SVG file import not allowed

kaimikael, 2025-02-11:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.2

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Select File→Import>Import 3MF/STL/STEP/**SVG**/OBJ/AMF…
Pick an SVG file.
Receive an error message that SVG files are not readable.
See the attached screen shots.

[image]
[image]
[image]

### Actual results

I am not allowed to import an SVG file.

### Expected results

I expected to be able to read my SVG file.  According to the release notes for 1.4.0.16 (https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-v1_4_0_16): 
“In Bambu Studio V1.4 we also introduced support for importing SVG files as a model object. This means that you can take **any** SVG vector file and load it into Bambu Studio to prepare it for printing without needing to convert it to .STL using other tools.” [my emphasis]

So, has the ability been removed and just happens to remain in the menu and dialogue, or are there unnamed restrictions on the contents of the SVG file? (Attached as well)

### Project file & Debug log uploads

[Empty.3mf.zip](https://github.com/user-attachments/files/18749568/Empty.3mf.zip)
[foo.svg.zip](https://github.com/user-attachments/files/18749566/foo.svg.zip)
[Logfiles.zip](https://github.com/user-attachments/files/18749567/Logfiles.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-02-12:
Hello, please update to version v1.10.02.64.
https://github.com/bambulab/BambuStudio/releases

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00