bambulab/BambuStudio #5936: GUI: overlapping texts in german language at the 'All Plates Stats' in the preview tab

LukasHill, 2025-02-11:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Slice any objects on multiple plates with a duration of over 1h with the language set to german and have a look at the All Plates Stats in the tab preview with the section "Geschätzte Zeit"

### Actual results

The text of the actual duration is overlapping with the plate description

### Expected results

there is anough space between duration and plate description or better adapt the location of the text by the actual length of the text before

### Project file & Debug log uploads

[tarot_cards_70x131.zip](https://github.com/user-attachments/files/18751755/tarot_cards_70x131.zip)

[image]

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-05-30:
Hello, we will fix this.

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If you have any further questions, we will analyze again

closed 2025-07-17 by Haidiye00