bambulab/BambuStudio #5945: Crashing after slicing

oullah, 2025-02-14:
### Bambu Studio Version

1.10.2.64 beta

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 7 7700X 8-Core Processor, 4501 Mhz, 8 Core(s), 16 Logical Processor(s)
Memory: 32GB
GPU: NVIDIA TITAN RTX

### Printer

Bambu P1S

### How to reproduce

1. Open 3mf 
2. Slice file

### Actual results

Bambu will either crash immediately or will crash after a little while after slicing. This happens with a lot of different models, some of them are really simple. 

### Expected results

Bambu shouldn't crash

### Project file & Debug log uploads

[G-Man Projects - Geometric Spinner - Core - Multicolour 1.zip](https://github.com/user-attachments/files/18803450/G-Man.Projects.-.Geometric.Spinner.-.Core.-.Multicolour.1.zip)

[debug_network_Fri_Feb_14_13_32_31.log.zip](https://github.com/user-attachments/files/18803535/debug_network_Fri_Feb_14_13_32_31.log.zip)

The dmp file is over 2GB so I cannot upload it

### Checklist of files to include

- [x] Log file
- [x] Project file

oullah, 2025-02-14:
Here is one more log where it crashed in the middle of slicing: 

[debug_network_Fri_Feb_14_13_53_08.log.zip](https://github.com/user-attachments/files/18803562/debug_network_Fri_Feb_14_13_53_08.log.zip)

oullah, 2025-03-02:
I cannot print two materials on the same plate. Crashes every time. Version 1.120.2.76

[bambu-studio.exe.5000.dmp.zip](https://github.com/user-attachments/files/19043608/bambu-studio.exe.5000.dmp.zip)

[crash_Sun_Mar_02_10_14_40_0_done.log](https://github.com/user-attachments/files/19043609/crash_Sun_Mar_02_10_14_40_0_done.log)

[debug_Sun_Mar_02_10_14_01_5000.log.zip](https://github.com/user-attachments/files/19043610/debug_Sun_Mar_02_10_14_01_5000.log.zip)

QingZhangBambu, 2025-03-04:
@oullah The dmp file is not compatible with version 1.10.02.64. I checked the latest version and could not reproduce the crash. Can you try the new version and check if it works? If it still fails. Please upload a new dmp file with the latest version