bambulab/BambuStudio #5948: Bambu Studio crashes on switching to Silent

ryanjhill, 2025-02-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Additional system information

Apple M1 MacBook Air 16GB

### Printer

Bambu Labs X1C

### How to reproduce

Start print on desktop Bambu Studio
Immediately after, go to Control panel with print option, adjust speed from 100% →  50% (Standard to Silent)
Seemingly does not change, clicking off the widget shows that 100% is still the speed
Try selecting silent again
Bambu Studio crashes

### Actual results

I dismissed dialogue, but it's Apple's standard pop-up that says something like "This application crashed unexpectedly"

### Expected results

1. The speed should have been updated in the UI immediately
2. Trying to adjust the speed again should not have caused Studio to crash

### Project file & Debug log uploads

[debug_Fri_Feb_14_22_22_26_21276.log.0.zip](https://github.com/user-attachments/files/18808972/debug_Fri_Feb_14_22_22_26_21276.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-17:
Thank you for your feedback. May I ask if you have encountered this problem in the previous version? Please provide the log and DMP file so that we can analyze the issue and solve it.

BoBoYu-1, 2025-02-17:
Thank you for your feedback. Please kindly provide the crash and DMP log files. The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue