bambulab/BambuStudio #5949: Overhang Detection and Path Calculation Issue – Incorrect Print Order for Overhang Walls

darth-pika-hu, 2025-02-15:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS 15.3.1

### Printer

Bambu Lab P1S, X1E

### How to reproduce

	1.	Open Bambustudio and load the test plate model using default settings.
	2.	Slice the model without making any manual changes to overhang or wall settings.

### Actual results

The overhang region is sometimes detected as an outer wall and is printed in the air, resulting in an incomplete or “hanging” line. In some cases, the printed order is such that the outer (edge) line is printed after the inner lines, leading to poor bridging and quality in the overhang.
When “Print inner first” is enabled, the sequence may appear correct (i.e., printing the edge line first), but the overall result is still suboptimal.

### Expected results

The slicer should correctly detect overhangs. For overhang regions, the path should be calculated so that:

- If a line is printed in the air (not connected to a solid region), it should either be automatically removed or merged with adjacent lines to form a continuous bridge.

- For overhangs, the line closest to the edge should be printed first, followed by the lines further inward, ensuring a solid and stable bridge.

- The overall print order for overhangs should yield a smooth and continuous bridge with no dangling or incomplete lines.

### Project file & Debug log uploads

[Projects.zip](https://github.com/user-attachments/files/18809152/Projects.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

QingZhangBambu, 2025-02-18:
The bridge will fail because it lacks support. Regardless of the sequence, it is likely to fail. You can try printing infill first to confirm this.

the solution is to add support part.

darth-pika-hu, 2025-02-18:
Of course I know I can add support, otherwise I would not post it as a bug here. The algorithms behind it are flawed. That is why I post it here.

QingZhangBambu, 2025-02-18:
I mean, for now, printing the bridge first is not enough. we need something like arc overhang algorithms.

darth-pika-hu, 2025-02-18:
> I mean, for now, printing the bridge first is not enough. we need something like arc overhang algorithms.

Okay. So "we" means Bambu Lab Devs like you or you and I? Do I need to take a look at the code and submit the changes?

QingZhangBambu, 2025-03-04:
Our teams are working on that. If you have a good idea, please feel free to submit commits or talk about it. You always have good ideas.

darth-pika-hu, 2025-03-04:
I think instead of spending time on thinking and writing complicated algorithms, it maybe more efficient to add detailed control to the Aux cooling fan or do some airflow simulation and improve the design of the aux fan. LOL

QingZhangBambu, 2025-04-02:
@KuestenKeks could you share your 3mf files?