bambulab/BambuStudio #5950: Fonts on Mac Version

wesiho, 2025-02-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.3

### Printer

Bambu Lab A1 Mini

### How to reproduce

Add Text to existing Plate

### Actual results

During the Font selection all Fonts are visible, correct previewed, and can be selected. Most of them then just use something as a standard Font. Some of them works correct.

### Expected results

I expected that all on my Mac installed Fonts can be used for Printing. One of these, for me important, Fonts is SLACKEY.TTF

### Project file & Debug log uploads

[Schildchen leer v1.3mf.zip](https://github.com/user-attachments/files/18809915/Schildchen.leer.v1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-17:
Hello, we have fixed this issue internally. You will need to wait until the next major version. so sorry

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

closed 2025-07-17 by Haidiye00