bambulab/BambuStudio #5951: Text shape not about to input Chinese characters

earthruby, 2025-02-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sequoia 15.1.1

### Additional system information

CPU: 3.6 GHz 8-core intel core i9
Graphics: Radeon Pro 580X 8G
Memory: 40GB 2667 Mhz DDR4

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Click on "text shape"
use Chang Jie to input Chinese characters in "Input text"
failed to input Chinese characters

I also tried to copy Chinese characters from other source and paste in the "Input Text" box, but failed

### Actual results

not able to input Chinese characters with Chang Jie

Paste function not working, nothing will be shown when pressing "Command + v"

### Expected results

should able to show Chinese characters when using Chang Jie
and if I press "Command + V", should paste what I have copied

### Project file & Debug log uploads

[debug_Fri_Feb_14_15_16_20_17744.log.0.zip](https://github.com/user-attachments/files/18811095/debug_Fri_Feb_14_15_16_20_17744.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-02-18:
Hello, our internal version has been resolved. Please wait for the next major version~

earthruby, 2025-03-10:
I have updated to 1.10.2.76, seems the issue still exists~

earthruby, 2025-06-03:
> Hello, our internal version has been resolved. Please wait for the next major version~

this issue has not been solved in the latest 2.1.0.59 (Mac) yet...

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~