bambulab/BambuStudio #5952: Purge Tower expanding in wrong direction (to outside of buildplate)

DevTKSS, 2025-02-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

Bambu X1C Carbon, but happens also if slicing for P1S

### How to reproduce

Slicing a multi color Print and putting the tower either on the left or right upper corner of the build plate

### Actual results

Preparation view before sending it to be sliced

[image]
just fine!

after slicing, its not printable because of the tower is going over the build plate!:

[image]

the sliced result is disabling this print to be printed althought the Preparations has all been (WITH SPACE!) inside of the buildplate just fine, so just the tower is causing this issue.

### Expected results

the tower should expand to the exact other direction as it has been all the time => Y- / to the lower side of the build plate, so its using space of the plate

### Project file & Debug log uploads

I am sorry, but this 3D model is actually under license, so publishing it would be illegal. Just change the expanding direction back to point to the inside of the buildplate as before.

I dont know any way to get a logfile from the bambu studio software so not attachable also

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DevTKSS, 2025-02-17:
I don't have a problem with the fact that it is expanding!
The issue is, that it is expanding to y+ instead of y- axis!

seanbo, 2025-03-24:
I have the same issue.

seanbo, 2025-03-24:
> [@DevTKSS](https://github.com/DevTKSS) Thanks for reporting this issue! The size of the wipe tower is related to the layer height and the amount of filaments used in each layer. We can only estimate the size of wipe tower before slicing, it is limited by the slicing process. Apologetically, there is currently no good solution. so please refer to the size after slicing.

The problem, as I see it, is where it is choosing to start printing the prime tower.  I am having the same issue.  If the prime tower started at the top of the build plate and printed toward the bottom, this would not be an issue.

[image]