bambulab/BambuStudio #5954: Using modifiers to limit Fuzzy Skin results in excessive addition to print time and filament used.

Ri22o, 2025-02-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Load the attached .3mf file. (Litho Frame w Foot_fuzzy_BBL)
2. Click Preview to slice the model.
3. Set Color Scheme to Filament.
4. Observe and note Total Time and Total Filament values.
5. Set Process to Objects.
6. Click the orange locks to revert the modifier cubes to Global settings.
7. Click to regenerate the slice.
8. Observe and note Total Time and Total Filament values.

### Actual results

In Step 4, with the modifiers to remove Fuzzy Skin active, the Total Time is 9h41m and Total Filament is 146.52g.
In Step 8, after removing the modifiers, the Total Time is 7h19m and Total Filament is 124.68g.

[image]

[image]

### Expected results

Enabling the modifiers should not increase Total Time or Total Filament, if anything they should reduce it. For reference, in Orca Slicer, enabling the modifiers to remove Fuzzy Skin decreases print time from 7h13m to 7h3m.

The modifiers are being used to prevent Fuzzy Skin from being applied to the interior faces of the model. However, while they do remove the Fuzzy Skin when enabled, they also add nearly 2.5 hours and 22g of filament to the print. Total Time and Total Filament should be similar with and without the modifiers and there is clearly an issue with how they are being handled.

### Project file & Debug log uploads

[Litho Frame w Foot_fuzzy_BBL.zip](https://github.com/user-attachments/files/18812638/Litho.Frame.w.Foot_fuzzy_BBL.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

QingZhangBambu, 2025-02-18:
modifier is generating extra walls. I will mark this feature for review and ask the product manager to look into it. The modifier for fuzzy skin should not create additional walls.

[image]

Ri22o, 2025-02-18:
I notice the same behavior when attempting to apply Fuzzy Skin to half of a block. It will cut the block in half, add walls, and apply Fuzzy Skin between the halves.

[image]

[image]

BrettThePark, 2025-06-23:
Please fix this BUG. This is not a "feature" addition, it is a regression.

QingZhangBambu, 2025-07-21:
The latest version now supports painting fuzzy skin and adding fuzzy skin using a modifier without generating separate walls. Please download and try it.

closed 2025-07-21 by QingZhangBambu