bambulab/BambuStudio #5955: AMS filament not loading

ArielBanco, 2025-02-16:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

11

### Printer

Bambu Labs X1-Carbon

### How to reproduce

Unable to load filament from AMS

### Actual results

Unable to load filament

### Expected results

Load filament

### Project file & Debug log uploads

[debug_Sun_Feb_16_11_27_59_7200.log.zip](https://github.com/user-attachments/files/18813432/debug_Sun_Feb_16_11_27_59_7200.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-17:
Thank you for your feedback. Could you record a screen shot for us to see the actual situation? This would be helpful for us to analyze the problem.

ArielBanco, 2025-02-17:
I have since sorted the problem

I am not sure why I have to repeat the same process more than 10 times to
finally get it to work.
It has happened both times I have loaded new filament.