bambulab/BambuStudio #5965: 发送打印任务时点击取消，软件崩溃

eezhaolei, 2025-02-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

[image]

### Printer

P1S

### How to reproduce

当前状态下，点击取消按钮，软件崩溃，但是打印机已经开始打印。

### Actual results

当前状态下，点击取消按钮，软件崩溃，但是打印机已经开始打印。

### Expected results

软件崩溃，但是打印机已经开始打印。

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-03-12:
感谢您的反馈。我们已定位到此问题，但我们发现这个现象是“程序未响应”，而不是“崩溃”。如果您确认是“软件崩溃”，需要麻烦您提供下dmp文件。

eezhaolei, 2025-03-12:
对对没有错，是程序未响应

| |
赵磊
|
|
15620988440
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2025年03月12日 09:36 |
| 收件人 | bambulab/BambuStudio ***@***.***> |
| 抄送人 | eezhaolei ***@***.***>,
Author ***@***.***> |
| 主题 | Re: [bambulab/BambuStudio] 发送打印任务时点击取消，软件崩溃 (Issue #5965) |

感谢您的反馈。我们已定位到此问题，但我们发现这个现象是“程序未响应”，而不是“崩溃”。如果您确认是“软件崩溃”，需要麻烦您提供下dmp文件。

—

感谢您的反馈。我们已定位到此问题，但我们发现这个现象是“程序未响应”，而不是“崩溃”。如果您确认是“软件崩溃”，需要麻烦您提供下dmp文件。

XinZhangBBL, 2025-03-12:
好的，此问题会在最近的版本修复

XinZhangBBL, 2025-04-29:
@eezhaolei  Please check if the problem is solved or not
https://github.com/bambulab/BambuStudio/releases

closed 2025-06-03 by XinZhangBBL