bambulab/BambuStudio #5973: Bottom support interface is generated unnecessarily and incorrectly identified as the top support interface when using multimaterial for supports and set the bottom interface layers to 0

YZcat2023, 2025-02-19:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab github releases, Bambu Lab Official website

### OS version

Win11

### Printer

Bambu Lab P1S

### How to reproduce

When using another filament for support interface, if the bottom support interface is set to 0 layers, which means the bottom contact surface is disabled, the slicer will still generate it. This unintended surface is conducted by the top contact layer settings, not the bottom layer parameters.(which should be already disabled.) The same error recurs on PrusaSlicer v2.9.0: [Bottom support interface generated when same layer has top support interface](https://github.com/prusa3d/PrusaSlicer/issues/14139). Since this creates much more waste and is not an expected behavior, I think it's a urgency waiting to get addressed.

### Actual results

[image]

[image]

These layers (labeled by the red circle) shouldn't be there for the bottom interface layers are DISABLED.

### Expected results

In the absence of multi-material support, the problematic interface is no longer generated.

[image]

### Project file & Debug log uploads

[toufa.zip](https://github.com/user-attachments/files/18869403/toufa.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-02-20:
@YZcat2023 Thanks for your report. 
In Normal Support, while support interface and support base do not use the same filament, it's considered to add a base layer to improve their adhesion. I'm sorry that it cannot be set to absolute zero for now in such case. and I suggest you try tree support which may provide better result.

YZcat2023, 2025-02-20:
> [@YZcat2023](https://github.com/YZcat2023) Thanks for your report. In Normal Support, while support interface and support base do not use the same filament, it's considered to add a base layer to [...]
Bro, it seems you didn't get it. Actually the support is the very filament that is used to print the part and i just set the interface to be another filament. Besides, all options which are used to adjust bottom interface failed to adjust the unwelcomed interface. If you switch to filament mode, you'll notify this:

[image]

YZcat2023, 2025-02-20:
@NothFie  Besides, I'm concerned that supports grown on a different filamentmay increase the risk of detachment.

NothFie, 2025-02-21:
@YZcat2023 I get it. When base part and interface part of support are in different filaments (here we dont care about the model, just concentrate on support) and number of bottom interface layer is set to 0, there is an “insurance” which generates this interface layer you dont want to have. Since number of bottom interface layer is already set to 0, this redundant layer is totally set from a preset, so "all options which are used to adjust bottom interface failed to adjust the unwelcomed interface".
For now I have to say sorry that you cant remove this layer by adjusting settings (may be adjustable in future versions), so I still suggest you try tree support which may provide better result.

YZcat2023, 2025-04-03:
The problem seems to be fixed in the latest version.

closed 2025-04-03 by YZcat2023