bambulab/BambuStudio #5980: Zoom Distance Decreases Over Time

jblenman, 2025-02-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 (22H2)

### Additional system information

CPU: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz   3.60 GHz
Installed Ram: 64GB
GPU: AMD Radeon RX 5700 XT

### Printer

Bambu Lab A1 w/ AMS

### How to reproduce

I'm not sure how best to reproduce other than coloring the details of models for a while and zooming in very close and zooming out and rotating the model to find more details missed by the Fill color tool.  Sometimes I will spend over an hour (or multiple hours) coloring a model (because of the limited coloring toolset) and the issue becomes most apparent over a long period of time, so it might be hard to reproduce in a reasonable amount of time.  I'm not sure if maybe you can log the zoom magnification and view position to see a deterioration in view distance over time when trying to go to maximum zoom.  Maybe you can see minute differences that are only noticeable to the user's eye after a long time.

### Actual results

It becomes harder to zoom into details after coloring a model for a while.  The maximum zoom magnification seems to deteriorate over time while working on a model with the coloring tools.

More detail:
I have been doing a lot of model coloring in Bambu Studio and I have noticed that the longer I work on a model, the less I can zoom into the model after a while.  When I first start up the application, I can zoom in very close and clearly see all the triangles that make up the model mesh.  As I zoom out and zoom in and move the model around, it gets to a point where I can zoom in very little and it gets harder to color details, particularly small triangles that that the Fill color tool seems to miss somehow.

### Expected results

I would expect the zoom magnification to be consistent at all times, regardless of how long I work on a model or how long the application has been running.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18914639/log.zip)

[Pikachu_in_Sweater.zip](https://github.com/user-attachments/files/18914640/Pikachu_in_Sweater.zip)

This is only the most recent project I have had the issue occur because it took a while to color.  It happens in any project that I spend a long time coloring.

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-02-22:
Hello, we have also encountered this issue internally, but it is very difficult to reproduce after restarting, so it is also difficult to accurately locate the problem. If you have good replication steps and let us know, we will fix them. Thank you for your feedback. We will continue to pay attention to this issue as well~

Toriv1976, 2025-03-27:
I signed up here on github because this issue drives me absolutely INSANE!! I don't understand why the zooming is soo terrible and limited in Bambu studio and like the OP, it diminishes to less & less zooming capabilities as I paint EVERY model!! Bambu is already limited on the zooming which drives me nuts as I can never zoom in close enough for very fine detailed painting, but when it diminishes even more as time goes on, it's EXTREMELY frustrating! Why do they have such amazing hardware, but can't get at least the painting abilities improved. It also drives me crazy that we can't use the height range painting in all directions, rather than just horizontal. How is this not an option? Would it be that hard to implement both better zooming along with being able to use the height range tool to paint even straight lines in any direction? It would make painting soooo much easier and quicker. Please, please PLEASE look into this and fix the zooming issues and allowing us to use the height range tool in any direction.

Haidiye00, 2025-03-27:
Hello, the new beta version has been released. Welcome to download and experience it~
https://github.com/bambulab/BambuStudio/releases

Haidiye00, 2025-03-27:
Hello, please use the latest version to confirm again,
and can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)
If there are clear operational steps, we will definitely analyze and solve them~

Toriv1976, 2025-03-27:
Does the beta version address the zooming issues? I will definitely download the screenToGif software and record it because it happens EVERY time I paint, driving me absolutely insane and has for the past couple years. I really wish it would also allow us to zoom in super close for fine detail as it would save soo much time & frustration painting models. It would also save a ton of time & frustration if you guys would possibly implement the height range tool to be used in any direction rather than just horizontal as stated in my original comment. It would make it SOO much easier when trying to get straight, even lines on models where you want them vertically or, diagonal etc. (Any direction)

Toriv1976, 2025-03-27:
Also, is there a way to log the one I'm working on now that has this issue? If so, could you please tell me how to upload it to you?

Haidiye00, 2025-03-27:
Hello, I haven't reproduced your problem yet. I suggest you reproduce the process and record it, and we will analyze it~

Haidiye00, 2025-06-13:
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59
Hello, please upgrade to the latest version and see if there are still any issues. Thank you