bambulab/BambuStudio #5985: CRASHED WHILE SLICING, CRASHED WHILE DOING ANYTHING, THIS PROGRAM SUCKS DONKEY DICK

pulloptc, 2025-02-22:
### Bambu Studio Version

NEWEST

### Where is the application from?

Bambu Lab Official website

### OS version

INDOWS 10

### Additional system information

THE PROGRAM CRASHES DOING SIMPLE TASKS.  

### Printer

P1P, P1S 

### How to reproduce

CLICK SLICE, OPEN, SAVE, MESH,....  ANY PROCESS CAUSES STUDIO TO CRASH

### Actual results

SLICE A FILE, IT CRASHED

### Expected results

DO ANYTHING, LIKWE SLICE, SAVE, OR OPEN AND THE PROGRAM CRASHES

### Project file & Debug log uploads

IT CRASHES

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-24:
Hello @pulloptc 
thanks for reporting this issue,
we are sorry for the crash issue on your computer

In order to resolve your issue as quickly as possible，could you help to upload the crash reports to us ?
you can find them following this guide
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
and please send as many as you can of these crash reports

pulloptc, 2025-03-04:
[log.zip](https://github.com/user-attachments/files/19067944/log.zip)

pulloptc, 2025-03-04:
Thats all the logs...  I'm beyond fed up with you program.  It sucks terribly.  I can functionally use Orca Slicer to do any of this but cant connect it to my printers.  Bambu Studio crashes when attempting to do anything but print files I download from makerworld.  It crashes when I attempt to open files I just fixed and worked on in Orca.  It just crashes.   The beta was so bad I had to delete it and restart everything..

pulloptc, 2025-03-04:
>   Hello @pulloptc
> thanks for uploading the logs
> we have released the formal version 1.10.2.76
> you can use this version instead of the beta version
> and from the logs, we can not found the crash dmp files
> please help to upload the dmp files(such as bambu-studio.exe.17068.dmp, in the log directory) next time
>
> Hello @pulloptc
> thanks for uploading the logs
> we have released the formal version 1.10.2.76
> you can use this version instead of the beta version
> and from the logs, we can not found the crash dmp files
> please help to upload the dmp files(such as bambu-studio.exe.17068.dmp, in the log directory) next time
> when you encounter crash again

That version is not stable either. You guys do not listen to what is said.. Sent from me

lanewei120, 2025-03-05:
Hello @pulloptc 

we only keep up to 30 log files in the log directory
and we can not find the dmp files(crash log) in your previous log.zip

so we request your help to upload the log again
as soon as you meet the crash with the new version

we want to solve your issue ASAP, 
but also we need enough information

thanks