bambulab/BambuStudio #5988: Bambu Studio not sending to Printer 100 percent every

Caltustion, 2025-02-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

ProcessorIntel(R) Core(TM) i7-10875H CPU @ 2.30GHz, 2304 Mhz, 8 Core(s), 16 Logical Processor(s)
32gb RAM
NVIDIA QUADPRO P620

### Printer

BAMBU LAB A1 MINI

### How to reproduce

I pressed send to plate and would just stall when downloading

### Actual results

It had recently updated the fans last night and then i came in the morning, it was doing this issue.

### Expected results

My expected results are to ensure that i can use my laptop to send things over.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [x] Log file
- [ ] Project file

Stanleyouk, 2025-02-23:
I have just had exactly the same issue….everything working ok last night but today I have the problem

Stanleyouk, 2025-02-23:
For information I am using Mac OS and P1S

Caltustion, 2025-02-23:
I dont think its a printer issue. Im assuming it's a Bambu Studio Issue

Caltustion, 2025-02-23:
The software is linking with the printer, so there is no issue with the connection

Caltustion, 2025-02-23:
> I've even done factory reset and reconnected printer also changed sd card
> but still same issue

Is this a forum or does this is get dealt with by the developers?

Caltustion, 2025-02-23:
Have you tried logging out and in on bambu studio. or resinstalling the software?

Stanleyouk, 2025-02-23:
I checked on BAMBU lab site and all systems were up...didn't check with external. Sites tho
Sorry no idea if this gats picked by devs

BoBoYu-1, 2025-02-24:
Hello, may I ask if it's possible to try again to see if it's feasible to send and print? If it's not possible, could you please provide the log for us to analyze?