bambulab/BambuStudio #5990: Cannot see bottom of settings screen

drmoses99, 2025-02-23:
### Bambu Studio Version

1.10.2.72

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.3.1

### Printer

P1S

### How to reproduce

Try and open settings window, cannot see bottom options under Develop mode

[image]

### Actual results

listed above.

### Expected results

Should be able to see or scroll down to see the bottom of the settings window so that I know I've reached all options.

### Project file & Debug log uploads

not related to a project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-02-24:
[image]
Hello, we have tested internally and found that it is normal. We are not sure if there is any setting that caused the slider to disappear? May I ask if the mouse scroll wheel is functioning properly? thank you

drmoses99, 2025-02-27:
Yes, mouse scroll function working normally (Apple Mouse for the imac).
1.10.2.76 version also just installed and same issue.  On a separate email
I was also asked the resolution, looks like it's at the default for the
imac.

[image: image.png]

drmoses99, 2025-02-27:
> Could you please tell me the resolution of your computer?

[image: image.png]

Looks like 2240 x1260, showing as the default for the imac.  I also just
downloaded/installed the latest version and it's still occurring.  1.10.2.76

Haidiye00, 2025-02-27:
Hello, it seems that your image was not uploaded successfully~

drmoses99, 2025-02-27:
Ah, it was just the screenshot of the resolution that was asked on the
other reply.  for my imac, it's set at the default 2240 x1260,

Haidiye00, 2025-08-13:
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85
Please use the latest version, thank you

closed 2025-08-13 by Haidiye00