bambulab/BambuStudio #5992: Account Login

salinius, 2025-02-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.6.1

### Printer

Bambu PS1

### How to reproduce

1. Open Bambu Studio
2. Click the Login/Register Button
3. Click one of the "login in with other accounts" or "system browser login"
4. Agree to Terms of Service
5. See "Bambu Studio quit unexpectedly" crash screen.

### Actual results

[image]

### Expected results

Bambu studio crashes and has to be re-opened.

### Project file & Debug log uploads

Was not working in any project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

salinius, 2025-02-23:
May be still a bug, but only on initial install. Reboot of machine, enabled log in to occur without incident.

BoBoYu-1, 2025-02-24:
could you help to upload the crash reports to us ?
you can find them following this guide
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
and please send as many as you can of these crash reports